YoyoMario/TiltBetaBuilds #5: Betaflight rates do not seem to compute properly.

sethrobin, 2019-04-15:
I used Apoc's rate calculator to convert my BF rates to RF and my RaceFlight rates plugged into Tilt just fine. However, entering my Betaflight rates directly yielded unexpected results, so I expect there's a math issue. Shown below are the Betaflight and Raceflight values that come close to matching my rates. I arrived at the BF rates through trial and error.

[image]
[image]

sethrobin, 2019-04-15:
Here are my standard rates.
[image]

YoyoMario, 2019-04-16:
@sethrobin Hey! Thannks for submitting this issue! We are avare, since default setting on the beginning was Race1, once we introduced KISS and BF their expo and super rates are switched up. You should try just swithing those values... I'll give it a try as of this moment right now

YoyoMario, 2019-04-16:
@YoyoMario Hey, just tried those values, something is off... is that velocidrone? It says "inspired by betaflight" are we even sure they are using the real calculations? I tried matching values with Air2, and ours are same as on Air2... I have to try DRL Sim yet to see if theirs is the same as ours

YoyoMario, 2019-04-16:
@sethrobin And something is definetly wrong with BF rates

closed 2019-04-16 by YoyoMario