YoyoMario/TiltBetaBuilds #6: Dashboard hangs when selecting Betaflight as the rate type.

sethrobin, 2019-04-16:
Dashboard hangs when selecting Betaflight as the rate type. 
I have seen this when accidentally entering whole numbers while in BF mode as well. After entering an initial value, if I want to change it, I have to explicitly enter the 1's digit. Such as: "0.62" instead of ".62" because it will ignore the "." without the leading zero and take "62" as the entry, which sometimes locks up the modal window.

[image]

[Player.log](https://github.com/YoyoMario/TiltBetaBuilds/files/3086602/Player.log)

sethrobin, 2019-04-16:
Video link not working. Stand by.

YoyoMario, 2019-04-16:
Yup, just wanted to DM you about that

sethrobin, 2019-04-16:
Link updated http://cloud.sethrobinson.net/3e784f6a11f1

closed 2019-04-18 by YoyoMario