YoyoMario/TiltBetaBuilds #7: Sliders for FOV and Cam tilt angle won't stop when changed

simgirl, 2019-04-18:
The values will slide down until it reaches usually 18 or 0 whenever I try to adjust FOV and camera tilt angle.

YoyoMario, 2019-04-18:
@simgirl Thanks for that bug, it's related to the throttle at the moment, will be fixed once I switch from physics to UI! 👍

YoyoMario, 2019-04-24:
@simgirl  hey, is this still happening even with the latest builds? someone texted me that it's gone but I haven't touched that issue yet, can you confirm?

simgirl, 2019-04-24:
> @simgirl hey, is this still happening even with the latest builds? someone texted me that it's gone but I haven't touched that issue yet, can you confirm?

Hey, I'm not free to get back on the sim at the moment since I have work tmrw morning, will get to it first thing I get back, if that's okay? Thanks :D

YoyoMario, 2019-04-24:
@simgirl Hey, ofcourse! I work on this in my free time aswell, we're obivously different time zones tho! :)
You can also text me on discord after you fly a run, cheers!

YoyoMario, 2019-04-28:
@simgirl still an issue, right?

simgirl, 2019-05-01:
@YoyoMario Yes, still the same in the newest build. I've screen-captured my latest session on the latest version. Where can I post it for your reference?

YoyoMario, 2019-05-01:
@simgirl  you can message me directly on discord or post in video group with the issue :) thanks!

YoyoMario, 2019-05-10:
@simgirl  this should be fixed now, ill close this issue :)

closed 2019-05-10 by YoyoMario