YoyoMario/TiltBetaBuilds #17: FrSky Simulator Dongle only giving partial axis commands

H4mer, 2019-04-25:
Trying to setup the joystick menu using a FkSky Simulator dongle, with my transmitter setup for TAER, I am unable to get the joystick wizard to read each of 4 axis as a whole axis. My only options are X_Left, X_Right, Y_Right, and Y_Left, all of which make the sticks movement display jump around.

YoyoMario, 2019-04-25:
@H4mer can you record me a video? that's the point of calibration and other steps... should fix that issues if your input is not going from -1 to 1... what build are you running, mac or win? Also could you record a video for me

YoyoMario, 2019-04-25:
@H4mer hey?

H4mer, 2019-04-25:
Sorry for late response. I will setup my phone to record me using the joystick wizard to show the calibration issues. Hopefully I'll have a video available by tonight.

YoyoMario, 2019-04-25:
@H4mer Awesome! Thank you! :)

H4mer, 2019-04-25:
https://youtu.be/tN9A0JbDsrM
This is an unlisted video so it shouldn't get seen, but I will delete it from my channel after you've had some time to review.

YoyoMario, 2019-04-26:
@H4mer awesome video, hmmm we had one guy reporting that his dongle works, i see you have a "SIM" profile setup on your controller, can you try wiring it? to see if its only a dongle issue?

Just checking, that is a profile setup as from the youtube video we have pinned?

YoyoMario, 2019-04-26:
@H4mer  for some reason your -1 to 1, is when you throttle from 0->1... thats weird...

sethrobin, 2019-05-16:
FWIW, I'm on Mac OS and the FRsky dongle is working well.

YoyoMario, 2019-06-04:
@H4mer  New update is live with input totally re-done, shoudl work now

closed 2019-06-04 by YoyoMario