YoyoMario/TiltBetaBuilds #24: Throttle below 0% registeres as full throttle

bleakgadfly, 2019-05-07:
I have a bit of throuble formulating this in a good way, so bear with me:

On previous builds Ive been running the same profile on my Jumper T16 as Im using on Liftoff and they worked fine. However, on the new build anything below 0%/1500us throttle is being read as 100%/2000us so the first thing that happens when I start the game is that the quad takes off to the sky. It seems like negative values are being read as full throttle. Ive gone through Joystick setup/calibration multiple times and on step 4 the throttle is being read as "(-0.98, 2.0)" but this does not seems like it matters. Let me know if you would rather want a video than screenshots.

0%/1500us throttle on T16 - 0% throttle read in Tilt:
[image]

-100%/988us throttle on T16 - 100% throttle read by Tilt
[image]

4%/1521us throttle on T16 - 4% throttle read by Tilt
[image]

If I go to setup Joystick I see the same behavior. Here it reads throttle 0.0079, i.e. mid-throttle, while in reality it should be 0.5, since the controllers throttle is 75%/1757us throttle on the stick.
[image]

YoyoMario, 2019-05-07:
@bleakgadfly Hmmm, throttle in the middle should give a value around 0.0, so thats good, on throttle stick down, there should be a value around -1 and with throttle stick up should read around 1.

Do you think we could discord screenshare or teamviewer? I would like to see the readings that you're getting. If I understood previous Tilt builds worked? Or were you refering to Liftoff builds?

bleakgadfly, 2019-05-30:
This issue can be closed. This was caused by the channels being weight 100% and offset 0%, changing to weight 50% and offset 50% fixed this issue.

YoyoMario, 2019-05-30:
@bleakgadfly thanks for the update! :) Best regards!

closed 2019-05-30 by YoyoMario