YoyoMario/TiltBetaBuilds #26: Decimal entry in the dashboard is janky.

sethrobin, 2019-05-16:
Entering any number below 1, like 0.35, is difficult. There is a default ".1" that gets added as soon as you type the zero. The 1 digit has to be manually deleted from the keyboard once the other numbers have been entered as there is no other way to replace or delete it.

Best if you just see the video for demo. The first part of the video shows this behavior.

http://cloud.sethrobinson.net/fc309eb22b2c

sethrobin, 2019-05-16:
This has been a issue for a couple of weeks. Just getting around to logging it.

YoyoMario, 2019-05-16:
@sethrobin  thats a good point! also you can now navigate through the inputs by pressing "Tab" on your keyboard

YoyoMario, 2019-05-18:
@sethrobin still clanky but works better now

closed 2019-05-18 by YoyoMario