YoyoMario/TiltBetaBuilds #28: Throttle appears to have a curve applied

sethrobin, 2019-05-25:
Here's a NEW video showing my throttle input vs what I'm seeing in Tilt and on the Taranis. The Joystick Show window on the left is raw input. Also, the throttle pops to "not center" when the Tilt application window loses focus. I'm not above making stupid mistakes, but I believe I've ruled out operator error.
http://cloud.sethrobinson.net/e9a0b723251f

Also, here's an earlier video showing the window focus throttle behavior:
 http://cloud.sethrobinson.net/2487afbca1a2

YoyoMario, 2019-05-25:
no curve on throttle -> http://prntscr.com/nt4xh9
curve on the throttle -> http://prntscr.com/nt4xng
I really don't see what confuses you? 😕

YoyoMario, 2019-05-25:
@sethrobin ^

sethrobin, 2019-05-28:
I found the source of the curve. With very early versions I was sending the 0 to 100 throttle rather than -100 to 100, as advised by the setup video. At some point I forgot to turn that on when calibrating a new version and things just worked, so I thought you'd eliminated that requirement. If I use the full range of the throttle, the weird curve shows up. If I use 0 to 100, it's linear. What's crazy is I had been using this for quite some time without any other indication that something was amiss.

YoyoMario, 2019-05-28:
I got confused from messages between you and Gil on discord and this message, so is it working now?

YoyoMario, 2019-05-28:
@sethrobin ^

closed 2019-06-04 by YoyoMario